openfpga-library/analogue-pocket #16: Feed additions - core's full name & release info

neil-morrison44, 2023-02-26:
Just taken a proper look at the feed & there's a couple of things I've noticed which'd be nice to have:

- The core's full name in there somewhere for in-app links etc, (e.g `jotego.jtvigil`) - it's in with the `category` items just now but it's split up & I'm not sure what the logic is there since PC Engine CD has
  ```
  <category term="feed"/>
  <category term="Mazamars312"/>
  <category term="PC Engine CD"/>
  <category term="PC"/>
  <category term="Engine"/>
  <category term="CD"/>
  ```
  (if it's always the 2nd & 3rd category items that's fine too, just need to make sure)
- On `POST_TYPE_UPDATE` (if it's a github release'd core) it'd be nice if either:
  - the `content` was the body of the release notes
  - there's a link out to the release on github somewhere
  
  whichever's easier - to differentiate between updates incase there's a few of them for a single core grouped up

neil-morrison44, 2023-02-26:
Oh actually - it'd be nice to have a link out to the core's page (on github or wherever, if it exists) on the `POST_TYPE_NEW` ones too

joshcampbell191, 2023-02-26:
I fixed the issue regarding the extra categories for the PC Engine CD. It should now only list the author and the shortname (pulled from the `core.json`).

I was thinking of re-using the URL attribute from the `core.json` instead of relying on the repository URL since this would be a more generic approach.

neil-morrison44, 2023-02-26:
the URL from the core's json works - I think most core devs have been setting that

Is there a way to programmatically label each category? I'll just read it as `["feed", author, short_name, type]` in order but if there was a way to say what each is it'd be a bit more future proof

What're your thoughts on getting some sort of release notes / link to release notes in there? it probably would end up as an optional thing since it's never going to work for the non-github release cores (plus some cores have a _lot_ of text in their release notes)

neil-morrison44, 2023-02-26:
[image]
This is how I'm planning on presenting the feed by the way, it might appear elsewhere in the UI too but I'm not 100% sure yet & depending on what's feasible there might be a link in there or clicking it brings up a modal or whatever

joshcampbell191, 2023-02-26:
> the URL from the core's json works - I think most core devs have been setting that
>
>
>
> Is there a way to programmatically label each category? I'll just read it as `["feed", author, short_name, [...]

Unfortunately that's now how categories work. I ordered them this way to keep cores by the same author together rather than putting all the posts in the same folder.

My only issue with release notes is that it only applies to GitHub releases which is why I went with the contents of the `info.txt`. We can look into it though.

joshcampbell191, 2023-02-26:
I could about re-using the author field of the post for the core author (from the `core.json`).

neil-morrison44, 2023-02-26:
Yeah - it's working as is, was just thinking incase something else comes along in the future that gets categorised differently.

I know what you mean with the release notes - it's not generic enough - it'd be nice if the `new` ones used the `info.txt` & the `update` ones either used release notes or fell back to the `info.txt` (would mean supporting both the markdown & the html though)

neil-morrison44, 2023-02-27:
That's a version of Pocket Sync up now that uses the feed API

I was going to wait a bit / add some more features (like showing the contents, notifications, etc) - but I noticed I'd not fully moved to V2 of the inventory & the app would request from V1 after 15 minutes and break everything...

Going to close this issue now since I'm able to get the cores full name & the release notes bit is on the radar at least

closed 2023-02-27 by neil-morrison44