openfpga-library/analogue-pocket #26: [Feature Request] Auto-Merge known-good PRs?

neil-morrison44, 2023-04-28:
Hey,

(I'm happy doing a PR with a workflow for this if it's wanted)

I think you can set GitHub to auto-merge PRs that pass checks & get approved and you can have workflows put approvals on PRs.

So you could have a workflow which places an approval on PRs based on criteria like:
- it only adds a core that's not yet in the system and doesn't change other files etc
- it's been raised by a known user (might be less important than the others & you'd have to maintain a list)
- the core is by a known author

Then the checks run to see if the core works etc & it gets auto-merged if everything looks good

mattpannella, 2023-04-29:
yeah thats a great idea. then no one even has to worry about babysitting the repo to keep an eye out for pull requests (by no one, i mean josh lol)

joshcampbell191, 2023-04-29:
That's a great idea! How about we go with the first and last one you proposed?

> * **it only adds a core that's not yet in the system and doesn't change other files etc**
> * ~~it's been raised by a known user (might be less important than the others & you'd have to maintain a list)~~
> * **the core is by a known author**

This should keep the reviews down to a minimum.

@neil-morrison44 Thanks for offering to submit a PR for this!

neil-morrison44, 2023-04-29:
@joshcampbell191 

Looks like the auto-merge stuff can't be used for users without write permissions (and doing a workaround for that is a little risky) so I'm going to try something a little different:

- Issue template including fields for username & a code block for yml e.g.

### Username:
spiritualized1997
### Core:
```yml
    - display_name: Spiritualized Adventure Vision
      repository: openFPGA-Adventure-Vision
```

- Issue template also some hidden stuff so that a github action that runs on each issue being created can notice it & parse out the username & core
- looks to see if the username already exists & that the core doesn't exist and adds it to the yml
- github actions programmatically creates a PR
- It either uses the auto-merge stuff or just waits till the checks complete

Should be fine & less risky since github actions remains in full control of the PR

joshcampbell191, 2023-04-30:
How about we go with the same format as the respositories.yml?

```yml
- username: agg23
  cores:
    - display_name: Arduboy for Analogue Pocket
      repository: openfpga-arduboy
```

This could allow users to specify multiple cores for different authors at the same time. Also, it simplifies the template to a single field.

neil-morrison44, 2023-04-30:
I think (at least for this first pass) limiting it to 1 core per issue makes sense - I discovered github's new Issue Form thing which makes it quite nice to gather multiple fields (it'd probably be possible to avoid having the user do any yaml at all but I've not gone that far yet).

My fork at https://github.com/neil-morrison44/analogue-pocket is now processing issues -> PRs -> approved PRs -> Merged PRs -> Closed issues

(the workflow code's a bit messy and I've removed most of the cores so it doesn't run into github's rate limiter but I'll fix that before a proper PR)

I'm not 100% sure it's not just working because the PRs are being opened with the repo owner as the author, so if you / @mattpannella could open an issue with a core in it to test that'd be handy

---

Assuming it all works there's a couple of steps to go through aside from merging in the workflow & form:
- Creating a branch protection ruleset
- Creating a github app to auth the PR in a way that allows the workflows to run & means the approval and the PR are from different "users" (https://github.com/peter-evans/create-pull-request/blob/main/docs/concepts-guidelines.md#authenticating-with-github-app-generated-tokens)

neil-morrison44, 2023-05-01:
I don't think I can have a repo fork that's both clean enough for a PR and that still does the auto-PR thing successfully - so give me a shout whenever you've given it a go @joshcampbell191 & I'll clean up my fork & raise a PR.

@mattpannella's given it a go here: https://github.com/neil-morrison44/analogue-pocket/issues/16 and it worked

joshcampbell191, 2023-05-01:
I'll try and get to it tonight. I appreciate your work on this!

joshcampbell191, 2023-05-02:
Just took a look at the workflow / script and was wondering if it would be possible to make the following change:
- When an **unknown** author is detected, a PR is submitted but not auto-approved.

Also, it would be good to add logging when those scenarios (unknown author / existing core) are encountered.

Otherwise, it looks good!

neil-morrison44, 2023-05-02:
That's it doing a bit more now - it'll open but not approve PRs from unknown authors & it feeds back to the initial poster with a comment saying what's going on for unknown author / existing core / invalid yaml

joshcampbell191, 2023-05-02:
I'll test again shortly and assuming everything checks out, we should be good to go!

joshcampbell191, 2023-05-05:
@neil-morrison44 Here are the results after running through a few different scenarios.

- [X] [Test #1 - New Core Existing Author](https://github.com/neil-morrison44/analogue-pocket/issues/45): PR Created and Accepted
- [X] [Test #2 - New Core Unknown Author](https://github.com/neil-morrison44/analogue-pocket/issues/47): PR Created and comment added
- [ ] [Test #3 - Existing Core](https://github.com/neil-morrison44/analogue-pocket/issues/49): Comment added
  - Issue could be automatically closed
- [ ] [Test #4 - Invalid YML](https://github.com/neil-morrison44/analogue-pocket/issues/50): Nothing happened
  - I suppose this was because the YML was valid, but the adder failed to parse it
  - **Suggestion**: We'll most likely require multiple RegEx patterns to parse the core repository properties separately
- [X] [Test #5 - Invalid Repository](https://github.com/neil-morrison44/analogue-pocket/issues/53): PR Created and Failed as expected
  - **Suggestion**: A comment could be added to provide feedback when this occurs

Finally, when a error occurs, are submitters able to fix it by editing their issue? The triggers seem to suggest it will only do it on re-open. Does this mean we should default to closing the issue when an error occurs?

neil-morrison44, 2023-05-05:
Test 3 - yeah, this should be doable.

Test 4 - Yeah, Ideal world they'd all be separate form fields but with the different types of core that's a little hard with how limited the forms are (i.e. if you could show / hide form inputs based on prior answers it'd be easy). I think it's enough for a first pass as is, especially since it's a limited number of users currently submitting cores.

Test 5 - bit hard to do, once the PR's opened the workflow is hands off, would need a second workflow running on certain PRs. Likely not worth the hassle, user can click through to PR & see the failure.

Yeah I wasn't sure about setting it to `edited` since then they won't go through the form again. Truthfully the `reopened` thing was only in as a debugging speed up, might be better to only have `opened` & just require re-submission.

joshcampbell191, 2023-05-05:
Let's go with the fix for Test #3 and change the default behaviour when encountering errors (closing the PR).

Once addressed, we should be good for a PR. 

Thanks again!

PR #28 (2023-05-07): Add create-pr-on-issue workflow
Closes #26

Steps required to get this working:
- Add a branch protection rule
[image]
- Tick this setting
[image]
- Create a github app with these permissions & allow it to access the Repo & set up its secrets as the `PR_CREATOR_APP_ID` & `PR_CREATOR_PRIVATE_KEY` action secrets
[image]

(roughly the steps from https://github.com/peter-evans/create-pull-request/blob/main/docs/concepts-guidelines.md#authenticating-with-github-app-generated-tokens but it doesn't need to access the user list stuff)

Should be good to merge now, my local fork will be back to running [... cut]
closed 2023-05-07 by joshcampbell191