openfpga-library/analogue-pocket #243: [New Core]: Flappy Bird

mattpannella, 2023-07-06:
### Core Author username

somhi

### Core .yml snippet

```yml
display_name: Flappy Bird
repository: FlappyBird
```

joshcampbell191, 2023-07-06:
@mattpannella Looks like this core is missing some key files to complete the metadata. Will see about contacting the owner with the info.

mattpannella, 2023-07-06:
the platforms folder is lowercase on the release zip. I was thinking that's the issue. I'll make a PR later when I'm at a computer. can we just rerun the action after that or does a new issue need to be made

joshcampbell191, 2023-07-06:
We should be able to re-run it. I'll see about adjusting it to be case insensitive.

mattpannella, 2023-07-07:
https://github.com/somhi/FlappyBird/pull/1

neil-morrison44, 2023-07-07:
The core’s the right place for the change to be made - depending on your OS you might end up with both `platforms` & `Platforms` after unzipping it.

neil-morrison44, 2023-07-07:
I made this a while ago: https://github.com/neil-morrison44/openfpga-validator but no one’s really been using it so it’s missing some of the things we added for the pc engine core etc.

(Aside from JOTEGO since I put it into the CI flow there to avoid bad zips)

There could be some way to use it to exit early / get better error messages

somhi, 2023-07-07:
Sorry for platform's typo. Just accepted the PR and generated a new zip into releases.

closed 2023-07-07 by github-actions[bot]

mattpannella, 2023-07-07:
no worries i think that fixed it! @somhi

somhi, 2023-07-07:
How does that github-actions work @mattpannella ?

As you can see in the repo I have multiples ports, so in releases I'm publishing for different boards.

mattpannella, 2023-07-07:
oo good catch. as far as i understand it's just grabbing the latest release and assuming there's a pocket build in each release. 
gonna probably need to adjust the process again @joshcampbell191

joshcampbell191, 2023-07-07:
That's correct @mattpannella. Most if not all cores that support multiple platforms / chipsets distribute them either via different .zip files under a single release. I believe this method is standard practice from a release perspective AFAIK.

somhi, 2023-07-07:
Do you mean that for each new release I should always include the Pocket version?  I can do that as far as I don't forget it :)
There is also a checkbox for new releases to avoid marking them as the latest release, not sure if that is took into account.

somhi, 2023-07-07:
By the way @mattpannella I'm adding new platform image from dyreschlock  and changing name from flappybird to Flappy Bird. Should I change also  "category": "Others" to   "Arcade" ?