openfpga-library/analogue-pocket #247: Version missmatches

drpaneas, 2023-07-08:
Comparing with the actual individual project release, the following cores have wrong version/tag metadata:

```
Version mismatch: 0.1.0 != 0.1.5 for Exerion
Version mismatch: 0.1.0 != 0.1.1 for Amiga
Version mismatch: v3.1 - Sep. 24, 1962 != 1.1.0 for PDP-1: Spacewar!
Version mismatch: 1.2.0 != 1.3.0 for Game Gear
```

joshcampbell191, 2023-07-09:
@drpaneas Thanks for submitting the issue.

Please note that the version is pulled directly from the `core.json` file (https://www.analogue.co/developer/docs/core-definition-files/core-json#metadata) for each core. This version was chosen over the GitHub release as this is also the version displayed on the Analogue Pocket. It also allows for consistent versioning of the cores no matter the distribution method chosen by the developer.

Hope this clears things up.

closed 2023-07-09 by joshcampbell191

drpaneas, 2023-07-09:
@joshcampbell191 thanks for the clarification. I wasn't aware of this. Then I guess I have to open individual issues to the original repos, because their release pipeline seems strange/wrong.