openfpga-library/analogue-pocket #291: [Feature Request] Tracking replacement cores

neil-morrison44, 2023-08-02:
Hey,
Opening this as a way to keep track of the inventory features that'll make dealing with the replacement cores easier.

I believe what we've settled on (or at least what I've, barely, started looking at implementing support for) is this:
- Cores which want to tell us they replace another core will contain a `updaters.json` with a `previous` property
- which is an array of enough data to find the old core(s)
- It'll contain multiple items if the core's been renamed / transferred ownership multiple times, so even if a user skips a version they'll end up in the same place

```json
{
  "previous": [ { "author": "author_name", "platform_id": "platform_id", "shortname": "short_name" } ]
}
```

e.g. if the "Dominos" core was replaced the new core would have a file like...

```json
{
  "previous": [ { "author": "ericlewis", "platform_id": "dominos", "shortname": "dominos" } ]
}
```
(`shortname` & `platform_id` won't always be the same, and technically a core might have multiple platform_ids but if any one started doing that there would likely be bigger issues)

What would be helpful from the Inventory is a new property that exists on both the replacement core's info & the core(s) that've been replaced - just linking to the other in the `author.shortname` format.

e.g.
```json
{
  ...
  "replaces": "ericlewis.dominos"
  ...
}
```

```json
{
  ...
  "replaced_by": "ericlewis.dominos"
  ...
}

The `replaced_by` is more important for us, with `replaces` more of a nice to have to show _somewhere_ in the UI.
`replaces` could need to be an array incase the core's been replaced multiple times, not sure.

There is the possibility of circular / hostile replacements but... hopefully that doesn't happen.

On the updaters side we'll:
- Know not to show any core with a `replaced_by` in the lists of installable cores
- Show UI / CLI Messaging explaining what's been replaced by what etc
- Unrelated to the inventory response: if we spot a `updaters.json` with a `previous` we'll know to move files about on install so the new core gets the old core's Saves / Assets / etc

@mattpannella might have more things to add for the CLI specifically, but I think this covers everything Pocket Sync could want

joshcampbell191, 2023-08-05:
After reading through, I'm not entirely sure this should be a function for the inventory. Ultimately, it should be up to the user to decide which core they want installed. Wouldn't it make more sense to archive the original repository and update the inventory to point to the new repository? The original core could remain in the inventory and be flagged as archived (and reflected in the API).

neil-morrison44, 2023-08-05:
Yeah - I don’t think we’re wanting much more than that from the inventory, just that the flag that says the core’s been replaced says what’s replaced it.

It sounds like they’re not keen on doing it via archiving the previous repo - since that’d involve the previous developer doing something & wouldn’t support a core being renamed but staying within the same repo.

mattpannella, 2023-08-11:
sorry, havent been around, i've been busy so i've been mia. i guess ultimately, i don't have particularly strong feelings either way (archiving old repos, or implementing this new stuff), i'm just gonna implement whatever the devs want to make it easier for people to manage their stuff

and i guess, ultimately, if it doesn't get implemented in the inventory, we can always just parse the updaters.json file locally and work off that (as long as the old cores are removed from the inventory. otherwise i guess thats an issue haha)

neil-morrison44, 2023-08-11:
We can yeah - but without it appearing in the inventory we’d only know when / if we went to install the new core.

Say there’s a user that’s only opted in to a limited amount of cores - unless the inventory tells you that Core B replaces Core A you’ll never download Core B and see the updaters.json file.

Bigger issue in Pocket Sync since it’s all manual - there’ll be an update to a core sitting in the “available cores” without the user being told it’s a replacement / update of a core they use

joshcampbell191, 2023-08-11:
Sorry I haven't gotten to this yet. Been pretty busy lately. I'll try and put some time together to work on this and have it available in the API.

joshcampbell191, 2024-02-12:
Replaced by will come at a later date. I plan on putting some work into this project soon as things have quieted down on my end.

closed 2024-02-12 by joshcampbell191