openfpga-library/analogue-pocket #388: Amiga is out of date

rsn8887, 2023-08-21:
Amiga is already at version 0.1.1, but the file here still shows 0.1.0. Shouldn't it be automatic?

joshcampbell191, 2023-08-22:
> Amiga is already at version 0.1.1, but the file here still shows 0.1.0. Shouldn't it be automatic?

Thank you for reporting the issue. 

The version is determined by the contents of the `cores.json` as described in the [Analogue documentation](https://www.analogue.co/developer/docs/core-definition-files/core-json).

In the case of the Amiga, the `cores.json` hasn't been updated to reflect the GitHub release version.

You can confirm this by checking it out here: https://github.com/Mazamars312/Analogue-Amiga/blob/main/dist/Cores/Mazamars312.Amiga/core.json

joshcampbell191, 2023-08-26:
@rsn8887 After further investigation, it appears you're correct. The API still points to the previous release as it hasn't been updated. 

As a result, I opened a PR for the Amiga core to address the issue (https://github.com/Mazamars312/Analogue-Amiga/pull/26). In the meantime, I'll manually update the reference to the latest release.

joshcampbell191, 2024-03-27:
Closed as per the latest release of the Amiga core.

closed 2024-03-27 by joshcampbell191