openfpga-library/analogue-pocket #397: "Sponsor only" flag for JTBeta cores

neil-morrison44, 2023-09-01:
Hi,

To avoid folks installing the jtbeta cores then asking why they don't work it'd be nice if the API returned some sort of "sponsor-only" flag on them.

Sounds like we could look for a "JTBETA" file in the data.json to see which cores need it.

Then in the updaters we can look at that flag & direct people to whatever value is already set in the "sponsor" bit when they look at the core.

Will need separate methods for each sponsor-only dev but only JOTEGO is doing it so far.

mattpannella, 2023-09-01:
yeah sounds good to me. plus, realistically, this will probably only ever be needed for jotegos stuff anyway

neil-morrison44, 2023-09-01:
Oh, and it's probably worth not adding the new JT beta cores now in JTBIN to the inventory till we've got this sorted somehow (just to avoid dealing with all the questions)

neil-morrison44, 2023-09-02:
Oh - it was pointed out on discord that the cores also contain a fake `"jtpatreon"` platform id (after the real one).

Can see that causing issues (think it definitely would if it was the first one).

I think using the existence of the file in data.json is probably better, but it might be worth checking in with JOTEGO to see (& so it’s known that we’re reading it and doing something with it)

mattpannella, 2023-09-02:
oh no did we finally hit a vote with multiple platform IDs lol. I'm pretty sure I just hard coded all my stuff to take element 0 in that array cuz I didn't wanna think about it

mattpannella, 2023-09-02:
i'm wondering if it's worth asking/defining a place to explicitly tell the updaters which dataslot file is the drm key. this way instead of just hardcoding some special line to look for "beta.bin" or whatever, there could be something put in the updaters.json file that says the file in dataslot x, or with name y is the drm file that needs to be moved (and ignore it when checking archive.org for dependencies)

joshcampbell191, 2023-09-02:
I think it would be easier if it were centrally managed in the API. I doubt many core developers put much thought in how the updaters manage this.

neil-morrison44, 2023-09-02:
Yeah - I think if we can work with what JOTEGO’s doing now it’ll probably be for the best.

with the exception being if too many things break with the double platform id - I suspect that was put in as an attempt to put the key into `assets/jtpatreon/common/` which (I guess) didn’t work since that’s not the instructions.

saying that, if we’re wrong about how the jtbeta file applies to all current & previous cores & we need specific ones for specific cores we’d need to be told somewhere which jtbeta files to put where…

mattpannella, 2023-09-02:
woops sorry I meant to tell the code building the api list. so instead of checking for beta.bin or jtbeta it could be explicitly defined by the core dev. but yeah considering no one else is going to realistically do this it probably doesn't matter

neil-morrison44, 2023-09-03:
https://github.com/jotego/jtbin/issues/340#issuecomment-1704141082

looks like I was right about the jtpatreon platform being there for having the beta.bin in a single location - but Analogue will support it so I guess it’s real…

mattpannella, 2023-09-03:
ah there it is. well, i think if we could maybe treat the platform at index 0 as the "true" platform and any additional are extra and thus don't need corresponding platform json files maybe it simplifies it just slightly? I don't know haha

neil-morrison44, 2023-09-03:
I’m modifying the validator to specify that _at least one_ `platform_id` must exist.

Making the platform at 0 special is risky if multiple platform ids ever get used for their original purpose (whatever that was) where there’s multiple real platforms

mattpannella, 2023-09-09:
before I merge my pr for this, what are we gonna call the flag on the response? `sponsor-only`, `sponsor_only` etc

joshcampbell191, 2023-09-10:
Based other existing references, we should go with `sponsor_only`.

mattpannella, 2023-09-10:
ok cool that's what i figured

joshcampbell191, 2023-09-18:
@mattpannella @neil-morrison44 How you you feel about renaming the option to `requires_license`? I believe this is closer to the proper terminology in this case?

mattpannella, 2023-09-18:
fine by me

neil-morrison44, 2023-09-18:
Yeah, that’s alright with me too

mattpannella, 2023-09-18:
whenever you get that flag check implemented we can add all the beta cores to the inventory. I already gave retrodriven my changes so he can merge them into his stuff. this way we don't push the cores without all the updaters ready to flag them haha

neil-morrison44, 2023-09-18:
I’m still to put it into Pocket Sync - should have some time to put it in tonight hopefully

mattpannella, 2023-09-22:
can probably close this now haha

closed 2023-09-22 by joshcampbell191