openfpga-library/analogue-pocket #472: [Feature Request] Optional Wikipedia links

neil-morrison44, 2023-10-27:
Hey,

For a while now Pocket Sync's tried to put in a Wikipedia link using the core's name (to varying degrees of success) - realised recently that the inventory could have it as an optional thing defined alongside the display name in the YML & fed out through the API & the website.

Clients (and the website) could show the link for each core that has one & maybe even show the first section as a preview blurb (no idea what Wikipedia's API is like but it seems like something that'd be possible).

Think it could end up really nice & help folks understand what each core actually is, especially now they're getting more niche.

joshcampbell191, 2023-10-28:
This feels out of scope for the project since the current data exposed by the API is entirely based on the contents of the core definition files.

neil-morrison44, 2023-10-29:
I was figuring it'd work similarly to the `display_name`. In that it's, optionally, populated in `repositories.yml` rather than any other source - like:

```yml
    - display_name: CPS1
      wiki: en.wikipedia.org/wiki/CP_System
      repository: jtbin
      path: pocket/zips/jotego.jtcps1.zip

```

If added we'd need to do a pass over the `repositories.yml` adding the links where they exist then modify the template for PRs to mention it for new cores.

joshcampbell191, 2023-10-31:
Wouldn't it make more sense to have "home" pages for each core / platform instead?