openfpga-library/analogue-pocket #487: [New Core]: ASO/Alpha Mission

espiox, 2023-11-14:
### Core Author username

RndMnkIII

### Core .yml snippet

```yml
display_name: ASO/Alpha Mission
repository: PocketAlphaMission
```

davewongillies, 2023-11-15:
> @espiox It doesn't appear that this core is currently packaged for the Analogue Pocket. It'll have to be fixed either on the developer's end or re-hosted elsewhere.

@boogermann was working with @RndMnkIII to get that fixed: https://discord.com/channels/834264850230018058/834286368565231647/1174006136312180847

RndMnkIII, 2023-11-15:
What exactly is missing or needs to be fixed?

closed 2023-11-15 by github-actions[bot]

joshcampbell191, 2023-11-15:
@RndMnkIII Everything looks good now. Yes, it was related to how the release is distributed via your repository. 

The inventory currently supports two methods:
1. A release artifact (zip) comprised of the files / folders expected by the Analogue Pocket
2. A zip that's hosted as part of the source code (ex. a dist folder)

Hope that clears things up! Also, it appears that your core is now packaged correctly and should appear in the inventory / updaters shortly. Congratulations on the release!