openfpga-library/analogue-pocket #493: [New Core]: SEGA System 16A

espiox, 2023-11-17:
### Core Author username

jotego

### Core .yml snippet

```yml
display_name: SEGA System 16A
repository: jts16
path: pocket/zips/jotego.jts16.zip
```

espiox, 2023-11-17:
Whoops, I think JT’s system broke and didn’t build the zip yet.

davewongillies, 2023-11-17:
> Whoops, I think JT’s system broke and didn’t build the zip yet.

Looks like its because vulgus has been removed: 

https://github.com/openfpga-cores-inventory/analogue-pocket/actions/runs/6907256161/job/18794017783?pr=494#step:7:8

```
/opt/hostedtoolcache/Ruby/3.1.2/x64/lib/ruby/gems/3.1.0/gems/octokit-4.25.1/lib/octokit/response/raise_error.rb:14:in `on_complete': GET https://api.github.com/repos/jotego/jtbin/contents/pocket/zips/jotego.jtvulgus.zip: 404 - Not Found // See: https://docs.github.com/rest/repos/contents#get-repository-content (Octokit::NotFound)
```

joshcampbell191, 2023-11-17:
Closed as per #497

closed 2023-11-17 by joshcampbell191