openfpga-library/analogue-pocket #564: Super GameBoy 2

hallem, 2024-02-06:
Since the Spiritualized SGB core can support both the Super GameBoy and Super GameBoy 2 roms, it would be great to have the Super GameBoy 2 rom added to the archive so the tools could give users an option on which one they wanted or create a second core for it that uses it.

joshcampbell191, 2024-02-12:
@hallem Thank you for reporting this issue. Unfortunately it would be up for the core developer to publish multiple cores for the Super GameBoy to target the individual ROMs. 

I wonder if it's possible to have both ROMs included but have neither match the expected name (`sgb_snes.smc`) in which case, I believe it should prompt you to choose one. Another alternative would be copy the existing core and modify the `shortname` in the core.json to `SuperGB2` and rename the Cores / Assets folders accordingly.

 Feel free to test this out and post the results.

hallem, 2024-02-12:
@joshcampbell191 So the existing Super GB core can support either the SGB 1 or SGB 2 roms with no issue. Creating a second core for the SGB 2 rom is really straight forward as well. I've got a few different setups for them on my [GitHub](https://github.com/hallem/openFPGA-Super-GB) based on playing with it a bunch. It's not too different from what dyreschlock did with Pocket Extras.

It was actually Pocket Extras that gave me the idea to reach out to see if we could get the SGB 2 rom added to the inventory archive as `sgb2_snes.smc` since the additional roms for Pocket Extras are included there. I'm in the middle of adding support to automatically download and apply any of the Pocket Extras to pupdate as well. With the SGB 2 rom in the archive, it would basically allow the same thing to happen.

The ultimate goal is to provide an automatic way to download the files that need to be changed on the existing SGB core to create a second core for SGB 2 so it doesn't have to be done manually.

What do you think?

closed 2024-02-26 by hallem