openfpga-library/analogue-pocket #573: "Extra" repositories

mattpannella, 2024-02-13:
This isn't so much a request for a feature on the inventory (though it may include one) but I'm just using this as a common discussion area

me and @hallem were discussing adding some support for @dyreschlock 's "pocket-extras" stuff in pupdate, and I started thinking it would make a lot more sense to come up with some kind of way for anyone to host pocket related files on github and let users grab that stuff via an updater, without needing us to build custom support and dependencies on their repos

i'll use the "pocket-extras" repo as an example. right now he basically has customizations and additions for a few existing cores. and they're broken into release zips. one for the donkey kong core (it adds support for a couple additional games), one for a bunch of jotego cores (additional games added to the cps1 and 2 cores. maybe a few other things), and something for the toaplan core
each zip is packaged like a regular distro package. it has Assets, Cores, etc whatever folders so you can just unzip directly to the root of your sd card and be good to go

soooo, i was thinking we come up with some format, similar to the updaters.json idea, but for github repositories. updater-extras.json or something. i'm not sure we would even need much, but something roughly along the lines of this:
```json
[
    {
        "title": "Donkey Kong",
        "description": "The Donkey Kong and Radar Scope cores were created by Eric Lewis, and he developed them to load a single rom at a time.  When you select the core, it'll immediately boot into Donkey Kong. Modern arcade cores allow you to choose a JSON file for the game you want to play, which allows the core to play multiple games, rather than just one.  Donkey Kong has many many romhacks, so I made some edits to the `data.json` which allows JSON to be selected.",
        "filename_filter": "pocket-extras-dk"
    },
    {
        "title": "Jotego Extras",
        "description": "Included in `pocket-extras` are several games that work on the release including: CPS1 and CPS1.5 Conversions to CPS2, including The Punisher, Street Fighter II Championship Edition, and Final Fight 30th Anniversary Edition, CPS1 Conversions to CPS1.5, including Mega Man Power Battle and Street Fighter Zero, Arcade_Offset hacks for CPS2 and CPS1",
        "filename_filter": "pocket-extras-jotego"
    },
    {
        "title": "Toaplan Extras",
        "description": "Toaplan descrioption",
        "filename_filter": "pocket-extras-toaplan2"
    }
]
```
now we can build something into pupdate/pocketsync/whatever that lets users subscribe to any number of pocket extra repos they want, and we can dynamically build out menu items or whatever linking to each release artifact

and to tie it into the inventory, maybe a new api endpoint on here to list available repos (for people who want to participate obviously) and then we can also just show that list in the updaters for easy subscription

i think that kinda sums it up? discuss (or dont!)

cc @neil-morrison44

mattpannella, 2024-02-13:
oh wait, i had one more idea (really, this was @hallem 's idea)
if a specific "extra" is tied to a single core (like a tweak to one, along the lines of the donkey kong example) we could tie that back into the main cores endpoint and include an `extras` element on the core that links to the available extras on that core

espiox, 2024-02-13:
Sounds great, I’m in! This would be a good way to distribute the System 16 Complete core without adding it to the inventory proper and confusing folks.

neil-morrison44, 2024-02-13:
Sounds interesting, yeah initial thoughts are:

- is it worth considering a more “recipe” based approach for the combined cores, so that they can be re-created whenever one of the sub-cores are updated? Could see them drifting out of date & users getting confused. Maybe a special part of the updaters.json that describes how to put it together & which cores to watch for updates but 🤷🏻‍♂️ 

- I think I’d probably represent this in the “Fetch” part of Pocket Sync, showing the list of them if we end up putting the list somewhere & allowing custom zips to be applied.

hallem, 2024-02-13:
> is it worth considering a more “recipe” based approach

Yeah this was on my mind too when I was building support for Pocket Extras in Pupdate recently. Being able to reapply them easily and also remove them easily would be something we should take into account.

> if a specific "extra" is tied to a single core (like a tweak to one, along the lines of the donkey kong example) we could tie that back into the main cores endpoint and include an extras element on the core that links to the available extras on that core

Yeah being able to do this I think is key. Each one of the extras for the most part ties to a specific core or set of cores, so being able to tell that on the core itself would make things a lot easier for the tools.

I know something like this would be very useful for the additional Super GameBoy cores that I build for the Super GameBoy 2 rom and the Vaporwave ROM hack and their respective 8:7 aspect ration changes.

davewongillies, 2024-02-13:
> is it worth considering a more “recipe” based approach for the combined cores, so that they can be re-created whenever one of the sub-cores are updated? Could see them drifting out of date & users [...]

Just throwing out an brainfart idea on the recipe approach. This might make things more complicated (although give them more power) for people with "extras" repos but would potentially make it easier for you, but how about the "recipe" being (maybe optionally) defined in a recipe file in the "extras" repo itself instead of here?

neil-morrison44, 2024-02-13:
@davewongillies yeah I think we’d need to come up with the common format for the “recipe” file then they’d end up in the extras repo & other repos - then updaters would offer to apply them (including downloading cores they’re needed etc).

No idea what the format would be, but it’d need to support combination cores where the original cores are hidden & things like the Super Gameboy core 2 where the original core would stick around. Probably other things too

mattpannella, 2024-02-13:
> Just throwing out an brainfart idea on the recipe approach. This might make things more complicated (although give them more power) for people with "extras" repos but would potentially make it [...]

yeah, the my initial thought was every extras repo hosts their own list/recipes/whatever 
so its maintained by ethe repo owner and the updater can just point at that repo and get a list of what's available

hallem, 2024-02-13:
Yeah as I was building out the functionality for Pocket Extras as things are today, the main issue I ran into was around the Toaplan 2 combination core. The hardest part was the repo doesn't host the RBF_R files from the individual cores and there wasn't an easy, clean way to got and get them. In particular this was due to how the cores were stored by the authors since they don't use the github releases functionality. So whatever we come up with needs to try and solve that somehow so it leads to less "hard coding" of behaviors.

hallem, 2024-02-13:
> is it worth considering a more “recipe” based approach for the combined cores, so that they can be re-created whenever one of the sub-cores are updated? Could see them drifting out of date & users [...]

> Just throwing out an brainfart idea on the recipe approach. This might make things more complicated (although give them more power) for people with "extras" repos but would potentially make it [...]

> yeah, the my initial thought was every extras repo hosts their own list/recipes/whatever
so its maintained by ethe repo owner and the updater can just point at that repo and get a list of what's available

Just to make sure I'm following, the location / existence of that file would be discovered through the openFPGA-cores-inventory then correct?

mattpannella, 2024-02-13:
> Just to make sure I'm following, the location / existence of that file would be discovered through the openFPGA-cores-inventory then correct?

i mean i was just tossing out half baked ideas to flesh them out haha
the initial thought was every extras repo hosts some kind of index/recipe list/whatever in a standard place (root or something) and then the updater either gets a list of available repos from the inventory or you can just straight up have the user input a repo they want to subscribe to in the updater even. "add extras repo", updater makes a github api request for that file and displays the available packages

neil-morrison44, 2024-02-13:
@hallem think so, yeah - the “recipe” would need to refer to the other cores by identifier (e.g `agg23.snes`) then whatever’s following the instructions could reach out to the inventory to get it (if it’s not there locally already)

The inventory already worries about & handles how different cores are released & converts it to a single download url, so you wouldn’t have to worry about that

neil-morrison44, 2024-02-13:
Think it’s worth getting a definite list of all the types of extra we want to support, so far I think it’s:

- Combination Cores, combine multiple cores into one then remove the sub-cores
- Variant Cores, add / change files in an existing core and create a new core without removing the existing core
- Just extra _stuff_, adds things - like instance JSON files - to an existing core

(Variant is probably a bad name since it’s already in the Pocket JSONs, but it’s unused)

I think we’d need a file which says what type of extra it is & what the instructions to apply it would be, alongside a zip containing all the files for the extra

neil-morrison44, 2024-02-13:
Oh - and we would need an extra file in the combined / variant cores which says which versions of the individual cores went into them (and probably which, ideally also versioned, extra recipe was followed for it) so we’d know when to update it & re-run the recipe.

hallem, 2024-02-13:
> Sounds great, I’m in! This would be a good way to distribute the System 16 Complete core without adding it to the inventory proper and confusing folks.

@espiox I didn't realize this existed. What I did for Pocket Extras & Pupdate with the current state of things can work for this too. I'll play with it a little while we're discussing all this.

hallem, 2024-02-13:
> Think it’s worth getting a definite list of all the types of extra we want to support, so far I think it’s:

I think you're spot on here @neil-morrison44. The only thing I'd add is that the _extra stuff_ will likely also include modifications to the core and/or data JSON files too in order to make them work.

dyreschlock, 2024-02-13:
Just a note: when Boogerman releases his updated Donkey Kong core, the dk-extras in pocket-extras will no longer be necessary.

neil-morrison44, 2024-02-15:
> Something else we should add to the information about the extras is version. That way the tools can tell if there's a new version for it and apply it. If there isn't a version specified, we'd have to decided whether or not to update it or not anyway.

Agreed, that's what I was meaning by
> ...ideally also versioned, extra recipe was followed for it

Not sure what it'd end up looking like - especially because no one's going to want to do an individual GitHub release per "extra".

Maybe (just throwing ideas out) it's some sort of system where each repo contains a main `extras.json` at the root that lists all the extras that repo contains & their version numbers (and maybe "type" / what cores they apply to?)

And on the updater side we'd need to store the version number used _somewhere_. (An `installed_extras.json` that lists the same stuff as that extras.json + a link to the repo for each installed extra?)

Then to centralise it we'd just need to have a list of extras repos, and it'd be easy enough to add a WIP extra repo to the file locally to test things.

hallem, 2024-02-15:
> Not sure what it'd end up looking like - especially because no one's going to want to do an individual GitHub release per "extra".

So far both @dyreschlock and @espiox version the releases. The combination core from Pocket Extras doesn't change the version number on the core.json though, just on the release files in GitHub. The combination core for the Sega 16 tracks the version number from Jotego, along with the release on GitHub.

The alternate cores for the Super GameBoy that I have setup are kinda versioned, but those won't really change unless the underlying spiritualized core changes.

neil-morrison44, 2024-02-15:
Ideally it's an explicit version & done the same way for all of the extras. If a new core comes in which can go into a combined one, or a new json file is added to a JSON one, or something added / changed in any of the other ones, then the updaters can automatically rebuild it (or tell the user that there's an update they can rebuild). (or even just fixing a typo)

Then the updaters themselves will track the cores the extras were applied to - to know when they've been updated & the extra should be rebuilt.

Means whoever put together the extra won't have to update it every Friday when Jotego rolls out an update & only when they actually want to update / change something in it.

With the `extras.json` it'd also give a place to put sponsorship information similar to what happens with the cores on the inventory just now - which could be shown to the user alongside the name etc.

It's a bit of overhead but I think it's probably less than trying to work something out with the GitHub releases system, but there's probably also a neater way using neither out there