openfpga-library/analogue-pocket #633: [New Core]: MyC64

markus-zzz, 2024-04-03:
### Core Author username

markus-zzz

### Core .yml snippet

```yml
display_name: markus-zzz MyC64
repository: myc64-pocket
```

markus-zzz, 2024-04-04:
Thanks @joshcampbell191 I have put together a small script (yet to be commited once confirmed that it is ok) to automate this process for me. A MyC64-Pocket.zip is now attached to the release.

joshcampbell191, 2024-04-04:
@markus-zzz 

Thanks for including a zip containing the core files and folders.

It would appear that the 'assets' and 'platforms' folders were created in the wrong casing. Please rename these (**A**ssets and **P**latforms) to match those expected by Analogue OS and re-upload the asset.

markus-zzz, 2024-04-04:
@joshcampbell191 

zip-archive updated

closed 2024-04-04 by github-actions[bot]

dukestah, 2024-04-09:
Many thanks for this project, I'm more than happy to see the C64 coming to the Pocket!