openfpga-library/analogue-pocket #660: [New Core]:

rolandking, 2024-04-27:
### Core Author username

rolandking

### Core .yml snippet

```yml
cores:
      display_name: Athena SNK 1986
      repository: openFPGA-Athena
```

rolandking, 2024-04-27:
I had a bad name for the platform image file so the automated script failed - that's fixed now so can the checks be re-run please.

joshcampbell191, 2024-04-27:
@rolandking Thank you for your submission. 

After re-running the pipeline and reviewing the logs, it still errored out. It would appear that the `Platform` folder was created with the incorrect casing (`platform` instead of `Platform`). 

Once the folder is renamed, it should be fine.

rolandking, 2024-04-28:
Thanks - I just fixed this and re-uploaded as v1.1.1. I think the directory is “Platforms” (plural). I found my original template had this wrong, not sure why

closed 2024-04-28 by github-actions[bot]