openfpga-library/analogue-pocket #694: [Feature Request] List games in Arcade cores

davewongillies, 2024-05-23:
Many of the arcade cores have more than one game (or maybe have multiple variants of the same game). Since all of them have json files for each of the games it'd be cool if the inventory could parse those json files (especially over the [main human readable website](https://openfpga-cores-inventory.github.io/analogue-pocket/)) and then show the games that the cores are capable of playing.

joshcampbell191, 2024-05-24:
I'm currently re-writing the updater. I'll take this into consideration when I release.

joshcampbell191, 2024-06-05:
@davewongillies do you have an example I can use?

mattpannella, 2024-08-01:
just saw this issue, not sure if it's still being worked on
but basically i think you'd just iterate through every instance json file in /Assets/{platform_id}/{core_id}/* and just take the filename of each json file as a game the core supports?

davewongillies, 2024-08-15:
> @davewongillies do you have an example I can use?

Here's a couple examples:

* https://github.com/rolandking/openFPGA-Athena/tree/master/dist/Assets/_PLATFORM_/_CORE_
* https://github.com/jotego/jtbin/tree/master/pocket/raw/Assets/jtcps15/jotego.jtcps15
* https://github.com/obsidian-dot-dev/openFPGA-Gauntlet/tree/main/dist/Assets/gauntlet/obsidian.Gauntlet

Now that I'm poking around I'm thinking this might be cumbersome since every core organises things differently in the repo itself, so you'd need to download each core artifact zip file instead and parse the json files inside the zip. I guess you could do that parsing only when a core has been updated.

joshcampbell191, 2024-08-15:
I've already thought about how I can address this and it should be fairly straight forward.

Most arcade cores that support multiple games use `instance_json` along with an extension of `json`. It should be as simple as extracting the filenames from the JSON files in the appropriate folder (common or core) depending on `core_specific_file`.