openfpga-library/analogue-pocket #828: new updaters.json option for display modes

mattpannella, 2024-09-29:
this honestly probably only effects pupdate since it blanket overrides all display modes on each core but I figured I'd post here for feedback/consensus before adding anything 

@mcclure expressed wanting to opt out of pupdate's method of overriding all display modes on each core when using the "force all display modes" option so I figured maybe just a flag to be skipped by that process (and maybe also not allow manually changing them?)

maybe something like 
```json
{
  "lock_display_modes": true
}
```

so the updaters know to leave them alone 
@neil-morrison44 @hallem

neil-morrison44, 2024-09-29:
Wonder if it'd be enough / help to know what the cores default (unaltered) video modes were.

In practice I doubt many core devs would add the lock status, but if we know what the default video modes were we can either allow the user to skip ones that are already well populated or allow them to reset them.

hallem, 2024-09-29:
So if a core had display modes, pupdate will store that list in the settings json file before it does the overwrite based on what the user chose. I've been working on an easier way other than reinstalling the core to reset the display modes back, but haven't had a chance to finish it.

At the end of the day, the user can edit the video.json file until their hearts content, so I don't think a block of any kind is really worth the effort.

What I plan to look into for pupdate at least is a way to give the user a choice to just either add the missing display modes or just overwrite the list completely. That's fairly straight forward. The only challenge I can think of at the moment is going to be around the sort order of them. The user base seems to prefer the core specific ones at the top of the list vs any other way.

neil-morrison44, 2024-09-29:
If we did have it in the inventory instead of manually saved within pupdate at time of change then Pocket Sync could add a “reset” button using that data too & you would reset it correctly to the default if / when the core updates to include more video modes. (Also one less bit of data to keep)

joshcampbell191, 2024-09-29:
Honestly, this sounds out of scope for the inventory. Display modes are ultimately a user preference. A core has been designed with a display mode in mind and the option to change it is up to the user.

mcclure, 2024-09-29:
To be clear my use case here is
* I am distributing a core outside the inventory system, or not distributing it at all
* The core is not, or should not be, on the pupdate list at all
* I am told if pupdate does not have a core on its list, it can actually remove displaymodes
* I just don't want pupdate to delete the displaymodes I added on purpose

mattpannella, 2024-09-29:
@joshcampbell191 hey sorry I was just using this repo as a common discussion place lol
I don't think this belongs in the inventory either (except I guess if you start just dumping the entirety of updaters.json into the API response without parsing)

mattpannella, 2024-09-29:
> If we did have it in the inventory instead of manually saved within pupdate at time of change then Pocket Sync could add a “reset” button using that data too & you would reset it correctly to the default if / when the core updates to include more video modes. (Also one less bit of data to keep).
> 
> Would also allow for things like showing different text / UI for author-set display modes.

yeah i think this kidna stuff could be cool and a good reason to hust dump te entirey of updaters.json into the api response
the inventory doesn't need to worry about it, but we can have stuff added to it and immediately use it

davewongillies, 2024-09-30:
> To be clear my use case here is
> 
> * I am distributing a core outside the inventory system, or not distributing it at all
> * The core is not, or should not be, on the pupdate list at all
> * I am told if pupdate does not have a core on its list, it can actually remove displaymodes (reverts to the "generic" set)
> * I just don't want pupdate to delete by accident the displaymodes I added on purpose

Just putting this here since I was mobile while we were discussing it yesterday over on the fpgaming discord.

The generic list of modes is:

````
0x10	CRT Trinitron	
0x20	Grayscale LCD
0x21	Original GB DMG	Requires core response
0x22	Original GBP	Requires core response
0x23	Original GBP Light	Requires core response
0x30	Reflective Color LCD	Generic
0x31	Original GBC LCD	
0x32	Original GBC LCD+	
0x40	Backlit Color LCD	Generic
0xE0	Pinball Neon Matrix	
0xE1	Vacuum Fluorescent	
```

And the core specific modes which get removed when you have a "generic" core (for want of a better word):

```
0x41	Original GBA LCD	
0x42	Original GBA SP 101	
0x51	Original GG	
0x52	Original GG+	
0x61	Original NGP	
0x62	Original NGPC	
0x63	Original NGPC+	
0x71	TurboExpress	
0x72	PC Engine LT	
0x81	Original Lynx	
0x82	Original Lynx+
```

This was done as previously we were doing all modes but with the latest modes that got added to AnalogueOS we hit the limit that AnalogueOS allows in the list of modes.

joshcampbell191, 2024-10-07:
> @joshcampbell191 hey sorry I was just using this repo as a common discussion place lol I don't think this belongs in the inventory either (except I guess if you start just dumping the entirety of updaters.json into the API response without parsing)
> 
> stuff gets lost on discord too quickly and the Internet hates forums.

I think it would make more sense to properly deserialize / reserialize the `updaters.json` for backwards compatibility in the future.

hallem, 2024-10-07:
I spent some time this weekend playing around with things and was able to get something in place for being able to merge the user selected display modes with the core specified ones for pupdate. It'll use the same question pattern that the core selector uses currently and stores the answer in the settings. It gives the user the choice to either merge, overwrite, or ask every time they go to apply display modes through pupdate.

When a user makes modifications, pupdate will store the core specified list. I've now added the ability to reset the display modes back to that stored core specified list, without having to reinstall the core.

I've also added the ability to exclude a core for the `enable recommended display modes` option. Right now the server file will do this for the 3 GB display modes and the Spiritualized.GB core since it doesn't support them. But if you enable reading the `display_modes.json` file locally, the user now has a pattern to have pupdate ignore any core they want when applying display modes.

I still have to update the documentation, but the [PR](https://github.com/mattpannella/pupdate/pull/349) is here if you want to take a look.

mcclure, 2024-10-08:
@hallem , that's great and i think it would address my issue without needing the updaters.json format changed.

mattpannella, 2024-10-15:
@mcclure i released that the other day. so hopefully its all good now