openfpga-library/analogue-pocket #935: [New Core]: memtest

davewongillies, 2025-02-06:
### GitHub Username

dave18

### Repository Name

MemTest_Pocket

### Pre-release

- [ ] The GitHib release which contains the openFPGA core is labeled as a pre-release.

closed 2025-02-06 by github-actions[bot]

davewongillies, 2025-02-06:
Weird, the action [failed](https://github.com/openfpga-cores-inventory/analogue-pocket/actions/runs/13188972415) but the PR still got merged

davewongillies, 2025-02-06:
Oh I see what's happened. The core files are in a `dist` directory in the zip file

joshcampbell191, 2025-02-06:
> Weird, the action [failed](https://github.com/openfpga-cores-inventory/analogue-pocket/actions/runs/13188972415) but the PR still got merged

I noticed that with the last one as well. Another fix for the list sadly. On the plus side, I should have time to look into it this weekend.

As for this problem, the openFPGA folders should be in the root of the zip. They're currently in a subfolder of "dist".