openfpga-library/analogue-pocket #950: [API] Version 3.0

joshcampbell191, 2025-02-26:
@mattpannella @neil-morrison44 

I'm currently looking at making changes to the API to support multiple versions for the same core. This would allow updaters to fetch previous releases and allow users to rollback if need be / install an alternative version.

In order to achieve this, I'm planning on restructuring the API response (while also renaming some items to better align with the openFPGA definitions for consistency.

Here's a preview of what a response could look like:

```json
{
  "data": [
    {
      "id": "agg23.Arduboy",
      "repository": {
        "platform": "github",
        "owner": "agg23",
        "name": "openfpga-arduboy"
      },
      "platform": {
        "category": "Handheld",
        "name": "Arduboy",
        "manufacturer": "Arduboy",
        "year": 2016
      },
      "sponsor": {
        "github": ["https://github.com/sponsors/agg23"]
      },
      "releases": [
        {
          "version": "0.9.0",
          "date_release": "2022-09-03",
          "download_url": "https://github.com/agg23/openfpga-arduboy/releases/download/0.9.0/agg23.Arduboy.0.9.0.zip",
          "data": {
            "data_slots": [
              { "extensions": ["hex"] }
            ]
          }
        }
      ]
    }
  ]
}
```

The `releases` property will contain an array of `release` objects which are defined by the following the properties:
- `version`: Version of the release.
- `date_release`: The date the release was made.
- `download_url`: URL to download the version of the core.
- `data`: Data slot definitions as defined in the [data.json](https://www.analogue.co/developer/docs/core-definition-files/data-json)
- `updaters`: Updaters definition as defined in the `updaters.json`

Let me know what you guys think before I put more effort into setting this up.

mattpannella, 2025-02-26:
i wonder if there's any additional data that could be included to help with filtering the cores list, client-side. maybe some of the stuff from the `framework` section of core.json?
```
"sleep_supported": true,
      "dock": {
        "supported": true,
        "analog_output": false
      },
      "hardware": {
        "link_port": false,
        "cartridge_adapter": -1
      }
```
not sure i would use this stuff but might be useful for pocket sync to show in the UI what features each core supports?

joshcampbell191, 2025-02-26:
I do capture it however it's not currently exposed in the API. I can include it in the `release`.

Something like this:

```json
"releases": [
        {
          "version": "0.9.0",
          "date_release": "2022-09-03",
          "download_url": "https://github.com/agg23/openfpga-arduboy/releases/download/0.9.0/agg23.Arduboy.0.9.0.zip",
          "core": {
            "framework": {
              "version_required": "1.1",
              "sleep_supported": false,
              "dock": {
                "supported": true,
                "analog_output": false
              },
              "hardware": {
                "link_port": false
              }
            }
          }
          "data": {
            "data_slots": [
              { "platform": "arduboy", "extensions": ["hex"] }
            ]
          }
        }
      ]
```

neil-morrison44, 2025-02-26:
New response looks good to me, wasn’t sure about the `framework` bits but seeing `version_required` in there makes me think it’d be useful - since we could warn users ahead of time that a core is incompatible with their current OS (…presuming it ever gets updated again).

Deployment wise would this version & the current version coexist for a while to allow tools & users to switch?

joshcampbell191, 2025-02-27:
The plan is to keep both versions running for a bit until you guys migrate.

After some additional playing around, I've come up with the following format:

```json
{
  "id": "agg23.Arduboy",
  "repository": {
    "platform": "github",
    "owner": "agg23",
    "name": "openfpga-arduboy"
  },
  "platforms": [
    {
      "id": "arduboy",
      "category": "Handheld",
      "name": "Arduboy",
      "manufacturer": "Arduboy",
      "year": 2016
    }
  ],
  "sponsor": {
    "github": ["https://github.com/sponsors/agg23"]
  },
  "releases": [
    {
      "download_url": "https://github.com/agg23/openfpga-arduboy/releases/download/0.9.0/agg23.Arduboy.0.9.0.zip",
      "core": {
        "metadata": {
          "version": "0.9.0",
          "date_release": "2022-09-03"
        },
        "framework": {
          "version_required": "1.1",
          "sleep_supported": false,
          "dock": {
            "supported": true,
            "analog_output": false
          },
          "hardware": {
            "link_port": false
          }
        }
      },
      "data": {
        "data_slots": [
          {
            "name": "HEX",
            "extensions": ["hex"]
          }
        ]
      }
    }
  ]
}
```

Here's a brief summary of the changes:
- The `platform` property has been renamed to `platforms` to align with `core.metadata.platform_ids` in the `core.json`. 
- The `id` property has been added to `platform` items.
- The `release` structure has been modified to include top level properties to align with `core.json` and `data.json`.
- The `data_slot` (previously `asset`) structure has been updated.
- The `platform` property has been removed from `data_slots` items since it's become redundant.
- The `name` property has been added to `data_slots` items.

Missing / Potential Changes:
- I need to re-include the requires_license to `release` items.
- I'm considering moving `sponsor` under repository since the two are linked.
- I'm considering replacing the `owner` and `name` properties of `repository` with `url`.
- I could include the remaining properties under the `metadata` property (`platform_ids`, `shortname`, `description`, `author`, `url`).
- I could include the remaining properties under `data_slots`.

joshcampbell191, 2025-02-27:
Wouldn't that be solved by using the `id` property under the `platforms` items?

neil-morrison44, 2025-02-27:
Looks like the bit I’m thinking of is in the parameters bitmask https://www.analogue.co/developer/docs/core-definition-files/data-json#:~:text=core's%20Interact%20menu.-,%5B25:24%5D,-Index%20of%20platform

So once parsed we’d end up with something like

```json
"parameters": {                  
  "core_specific_file": false,
  "instance_json": false,
  "platform": "platform_id_for_file"
}
```

Only matters if the slot bothers specifying it, otherwise it’ll just be whatever platform is in the 0th position in `platform_ids`

joshcampbell191, 2025-02-28:
Got it. Thanks for the clarification. I must've overlooked this detail when I originally implemented it.

Here's the resulting JSON:

```json
{
  "id": "agg23.Arduboy",
  "repository": {
    "platform": "github"
  },
  "platforms": [
    {
      "id": "arduboy",
      "category": "Handheld",
      "name": "Arduboy",
      "manufacturer": "Arduboy",
      "year": 2016
    }
  ],
  "releases": [
    {
      "download_url": "https://github.com/agg23/openfpga-arduboy/releases/download/0.9.0/agg23.Arduboy.0.9.0.zip",
      "core": {
        "metadata": {
          "platform_ids": ["arduboy"],
          "version": "0.9.0",
          "date_release": "2022-09-03"
        },
        "framework": {
          "version_required": "1.1",
          "sleep_supported": false,
          "dock": {
            "supported": true,
            "analog_output": false
          },
          "hardware": {
            "link_port": false
          }
        }
      },
      "data": {
        "data_slots": [
          {
            "name": "HEX",
            "extensions": ["hex"],
            "parameters": {
              "core_specific_file": false,
              "instance_json": false,
              "platform_index": 0
            }
          }
        ]
      }
    }
  ]
}
```

I added the `platform_index` property to `data_slots` items in addition to the `platform_ids` array to the `metadata` property.

joshcampbell191, 2025-03-22:
I just published the latest version of the API (v3) note that you will have to make changes to your applications in order to read the new response formats.

With this release, I also published Swagger / OpenAPI documentation for the current (and all previous) version of the API. You can find it here: https://openfpga-cores-inventory.github.io/analogue-pocket/api/swagger/

closed 2025-03-22 by joshcampbell191

neil-morrison44, 2025-03-22:
👍🏻 I’m planning on putting some time into Pocket Sync soon & getting another version out, will switch over & look into showing UI for the new V3 things