openfpga-library/analogue-pocket #979: Platform JSON info missing in V3

neil-morrison44, 2025-03-30:
I've started moving Pocket Sync over to V3, hit a bit of a hurdle in that the platform info that was in V2 doesn't have a V3 equivalent.

In V2 it came in as 
```JS
      "platform": {
        "category": "Arcade",
        "name": "Space Invaders",
        "manufacturer": "Taito",
        "year": 1978
      },
```

on each core

In V3 I'd probably expect it to come in alongside `"data"`, `"core"`, & `"updaters"` within each release. Maybe, to support cores with multiple platforms it'd come in as an array like

```JS
"platforms": [
  {
        "category": "Arcade",
        "name": "Space Invaders",
        "manufacturer": "Taito",
        "year": 1978
  }
]
```

Pocket Sync uses the `platform.category` to allow filtering on uninstalled cores, as well as the info in it on the uninstalled core's info page

That's the only important bit that I've noticed is missing between V2 & V3

joshcampbell191, 2025-03-30:
The platform info was moved into its own API (https://openfpga-cores-inventory.github.io/analogue-pocket/api/v3/platforms.json). 

This is similar to how they're structured in the PocketOS.

neil-morrison44, 2025-03-30:
Ah yeah, I saw that but for some reason it didn't click.

Cool, can move ahead

closed 2025-03-30 by neil-morrison44